siemens/kas #131: dump --resolve-refs fails when using name

kwavnet, 2024-10-11:
running `kas dump --resolve-refs` with

```
header:
  version: 14
repos:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    tag: kirkstone-4.0.20
    name: foo
    path: sources/poky
    layers:
      meta:
```

fails with

```
Traceback (most recent call last):
  File "/.env/lib/python3.10/site-packages/kas/kas.py", line 205, in main
    kas(sys.argv[1:])
  File "/.env/lib/python3.10/site-packages/kas/kas.py", line 194, in kas
    plugin.run(args)
  File "/.env/lib/python3.10/site-packages/kas/plugins/dump.py", line 225, in run
    config_expanded['repos'][r.name]['commit'] = r.revision
KeyError: 'foo'
```

the same call without the `name` in the config

```
header:
  version: 14
repos:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    tag: kirkstone-4.0.20
    path: sources/poky
    layers:
      meta:
```

works absolutely fine.
Reproduceable on 4.x releases and the next branch

jan-kiszka, 2024-10-11:
@fmoessbauer

fmoessbauer, 2024-10-11:
@kwavnet Thanks for reporting. This should easily be fixable. I'll send a patch to the ML by today. If you like I can give you credit in a `Reported-by` tag.

kwavnet, 2024-10-11:
> @kwavnet Thanks for reporting. This should easily be fixable. I'll send a patch to the ML by today. If you like I can give you credit in a `Reported-by` tag.

@fmoessbauer 😊 not needed, I was just stumbling upon that issue (functionality isn't mission critical for me), so I thought better to report - so as long as a fix lands in master at any point I'm good

commit e3473697 (2024-10-19): fix: dump repos by key instead of name

In most cases the repo name is identical to the repo id. However, this
only holds if no explicit name is set. The dump plugin reconstructed the
flattened repo dict by name, which is wrong as it is not identical to
the input.

This is fixed by getting the repos with their keys (instead of just the
repos) and properly reconstructing the dict based on the keys.

Fixes: #131
closed 2024-10-23 by jan-kiszka