siemens/kas #135: Setting environment variables in project configuration

Jesseeee, 2025-01-18:
Hi,

I was wondering if there would be the possibility to allow setting kas env variables in project configuration?

My specific use case is that we have a bunch of projects, some of which require specific containers, and it would be great if we could specify the container to use, for development as well as for reproducibility afterwards, this is one of the only things that's currently not really captured in our configuration/output and has to be manually added to know which container is needed for a specific build/project.

I was thinking of being able to specify these variables in the configuration as there might be other ones that are interesting to set fixed e.g. DL_DIR, though this is mostly convenience.

Thanks for your consideration,
Jesse

fmoessbauer, 2025-01-19:
Hi,

setting environment variables in the project config is already supported using the `env` key.
However, I guess you are looking for something different: A way to express the env-vars used by `kas-container` (e.g. by providing an env file). This is not yet possible, but someone recently asked for a similar feature on the kas mailing list.

Can you provide a bit more details on your use-case? E.g. how your development workflow looks like?

Best regards,
Felix

fmoessbauer, 2025-01-21:
> On the one hand we have our own containers (IT certificates and whatnot, extra libs e.g. for Xilinx builds) and we have to work with different versions of these, having a system where we would immediately know and use the right container would be very helpful.

That means you do something like this, right?

```bash
export KAS_CONTAINER_IMAGE="special-image:4.6"
kas-container build project-config.yml
```

> Second is that once we release a software, we'd also like to keep track of how exactly it was built and in which environment, and of course a container is very good for keeping that under control, but we'd also like to keep this information somewhere as a build artifact and currently that's something we have to do on top of the kas build itself.

Did you already look into the provenance feature? This should already provide most of the data. If you execute the build in a well-known CI system like GitLab-CI or GitHub Actions it also adds the job id. https://kas.readthedocs.io/en/latest/userguide/build-attestation.html. We could also think of providing an extension point so that the build environment can inject additional data.

Expressing the `kas-container` version in the project config is somehow a chicken-egg problem. In case of GitLab CI usually people do not use the `kas-container` script at all but just let the CI run the `ghcr.io/siemens/kas:...` image, so a "wrapper" would not help as the image is already chose once kas starts.

Noting what was used during build is easy with the standardized provenance information, though. Together with a flattened kas config (`kas dump <my-config>`) it should be pretty easy to reproduce the build.

Jesseeee, 2025-04-02:
> > On the one hand we have our own containers (IT certificates and whatnot, extra libs e.g. for Xilinx builds) and we have to work with different versions of these, having a system where we would immediately know and use the right container would be very helpful.
> 
> That means you do something like this, right?
> 
> export KAS_CONTAINER_IMAGE="special-image:4.6"
> kas-container build project-config.yml

That is correct.
> Did you already look into the provenance feature? This should already provide most of the data. If you execute the build in a well-known CI system like GitLab-CI or GitHub Actions it also adds the [...]

I'm not sure I'm really using it correctly but I tried, and I got the impression it wasn't quite what I needed
 
> Expressing the `kas-container` version in the project config is somehow a chicken-egg problem. In case of GitLab CI usually people do not use the `kas-container` script at all but just let the CI [...]

That is true, but we're currently also supporting a different number of containers as we have some exotic builds that are stuck on older versions, and sometimes we wanted to port extra tools inside it and so on.

From my perspective something like SSH_DIR, NETRC and so on is evidently environment config, what's in there depends on the build host, your corporate environment, proxies and so on.

But the KAS_CONTAINER_IMAGE is a bit that last missing puzzle piece for me to make starting a build straightforward, now I got to provide additional instructions denoting what container image should be used during development on all my different projects, I could of course do this with an additional wrapper script, but that kind of defeats the purpose as this is exactly what I hope to avoid by using kas(-container), having a single tool that expresses all the necessary config

fmoessbauer, 2025-04-02:
> But the KAS_CONTAINER_IMAGE is a bit that last missing puzzle piece for me to make starting a build straightforward

That means, adding a top-level option `kas_container:ghcr.io/siemens/kas/kas:4.7` to the config which is parsed by the kas-container script would solve this? IMHO this is something we could add.

Jesseeee, 2025-04-02:
> That means, adding a top-level option `kas_container:ghcr.io/siemens/kas/kas:4.7` to the config which is parsed by the kas-container script would solve this? IMHO this is something we could add.

Honestly? If this could be added that would solve all the issues I'm currently having (well except for cert updating but one thing at a time :) ) in terms of straightforward cloning/building/reproduceability of projects.

If you could point me in the right direction of where I could add this I'd gladly put in the work myself.

fmoessbauer, 2025-04-02:
> If you could point me in the right direction of where I could add this I'd gladly put in the work myself.

Sure! Contributions are always welcome. To see how to contribute, please read https://github.com/siemens/kas/blob/master/CONTRIBUTING.md

I further recommend to bring this topic up on the mailing list as there more people are reading it. Code-wise, the following locations are relevant:

- `kas-container`: do it similar to how we check for `build_system`
- json schema: add it to the [`schema-kas.json`](https://github.com/siemens/kas/blob/master/kas/schema-kas.json)
- documentation: document it in [`project-configuration.rst`](https://github.com/siemens/kas/blob/master/docs/userguide/project-configuration.rst#configuration-reference)

closed 2025-04-02 by Jesseeee