siemens/kas #137: Adding a way from the menu plugin to build in a container.

JosephMillsAtWork, 2025-01-22:
The title kinda says it all. Looked at the source code a bit and looks like there is not(that I can find) a way to "Save and Build" from the menu while using a container. 

If there is just let me know and close said issue.

jan-kiszka, 2025-01-22:
`kas-container menu`? Or what else do you mean?

JosephMillsAtWork, 2025-01-23:
@jan-kiszka  hello again. My thoughts are as follows. 

I am not sure that I have tried what you suggested if it works and the "Save and Build" runs under the container then that is amazing and we are done. 

If the "Save & Build" button does not run in the container after end user selects said button then the issue is still valid.

My thoughts are if the kas-container can already save & build and it builds in said container close the issue and ill document my upstream code.

Again thanks as always and looking forward to closing this issue.

jan-kiszka, 2025-01-23:
`kas-container` runs kas commands, including `menu` and any chained `build`, inside one of the containers that the kas project provides. Give it a try. See also how we describe the steps to configure and build images this way, e.g. here: https://github.com/siemens/meta-iot2050/?tab=readme-ov-file#build-example-image

JosephMillsAtWork, 2025-01-23:
@jan-kiszka  You're the best I'm gonna close this now.

closed 2025-01-23 by JosephMillsAtWork