siemens/kas #138: Adding a "test" plugin

ajlennon, 2025-01-22:
Hi there,

I am just getting to grips with (the awesome!) kas. I may have missed this but I am working to implement support for CI with meta-java and I would like to run  Yocto tests. Assuming this isn't present, would it make sense to add another "kas test" plugin?

(I am assuming for now I can override `KAS_TASK` ?)

Cheers,

Alex

fmoessbauer, 2025-01-29:
Hi @ajlennon ,

you should be able to execute the tests with one of these commands:

```bash
kas build --target <image> --task testimage kas.yml
# in case the target is set via the project config, this should also work
kas build --task testimage kas.yml
```

With that, I don't think a dedicated plugin is really needed. But we're happy to receive documentation patches to make it clear that it is really that simple.

ajlennon, 2025-01-29:
Awesome thanks! Will take a look at that then :)