siemens/kas #140: Is it possible to add an interactive mode to kas-interactive and see the building logs as normally Bitbake shows?

jetm, 2025-02-05:
Hi,

`kas-container` is very useful for CI, but it's useful as well for a local environment where you want to build in a Linux OS that is incompatible with Yocto.

When building locally with `kas-container` I see the `NOTE: Unable to use interactive mode for this terminal, using fallback` message from Bitbake. I understand this is the default way for the CI environment, but it would be nice to change it with some option when using kas-container from your PC.

My guess is it has to do with how the container is used. I was thinking like passing down the `-i` flag, like `kas-container --runtime-args '-i' build` but that doesn't work

fmoessbauer, 2025-02-06:
Hi @jetm , if you start the kas container from a terminal you should get an interactive terminal (showing the bitbake progress bars). If you run `kas-container shell`, you even get a fully interactive shell to do things in the yocto environment.

- Which container runtime are you using?
- On what OS?
- Which shell?

Example output:

[image]

jetm, 2025-02-06:
Hi @fmoessbauer,

With `kas-container shell` is the same result, and it shows the same NOTE message about interactive mode.

OK, it works by default; maybe it has to do with the container runtime, as I am using `podman`.

- Container runtime: podman
- OS: Arch Linux
- Shell: zsh
- kas version: 4.7

jetm, 2025-02-06:
Hi @michaeladler,

I installed kas from[ AUR package](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=kas). Locally, I modified the PKGBUILD to install kas 4.7.

Terminal: kitty 0.39.1 created by Kovid Goyal
TERM: xterm-kitty

michaeladler, 2025-02-07:
I don't use the AUR package because it's incomplete (e.g. missing some dependencies) and also not needed if you're only interested in kas-container. You can just grab [kas-container](https://github.com/siemens/kas/blob/master/kas-container) and invoke it directly. I just tested kitty with zsh and it worked fine, so I'm not sure what you're doing wrong. FWIW, I have set `export KAS_CONTAINER_ENGINE=podman`.

jetm, 2025-02-07:
@michaeladler Thank you for pointing out that line. I check the `infocmp` and everything is OK in my Shell environment.
 
With `konsole` works, it doesn't work with `ghostty`. But I managed to make it work with `kitty` in my environment by upgrading the kas Docker image from Debian `bookworm` to `trixie`.

Even now working, the message `NOTE: Unable to use interactive mode for this terminal, using fallback` is still showing, so, that is something else.

I wish to know what exactly was wrong with my environment, but for now I am satisfied with this "solution."

[image]

`Dockerfile` changes to upgrade kas Debian Docker image. After building with `./scripts/build-container.sh` and making `kas` to use that image with `export KAS_CONTAINER_IMAGE=ghcr.io/siemens/kas/kas:next`.

```diff
diff --git a/Dockerfile b/Dockerfile
index c6af84d..6d25c1a 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -21,14 +21,14 @@
 # OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 # SOFTWARE.

-ARG DEBIAN_TAG=bookworm-slim
+ARG DEBIAN_TAG=trixie-slim

 FROM debian:${DEBIAN_TAG} AS kas-base

 ARG SOURCE_DATE_EPOCH
 ARG CACHE_SHARING=locked

-ARG DEBIAN_TAG=bookworm-slim
+ARG DEBIAN_TAG=trixie-slim
 ENV DEBIAN_BASE_IMAGE_TAG=${DEBIAN_TAG}

 ARG TARGETPLATFORM
@@ -107,7 +107,7 @@ RUN --mount=type=cache,target=/var/cache/apt,sharing=${CACHE_SHARING} \
     apt-get update && \
     apt-get install -y -f --no-install-recommends \
             binfmt-support bzip2 mmdebstrap arch-test apt-utils dosfstools \
-            dpkg-dev gettext-base git mtools parted python3 python3-distutils \
+            dpkg-dev gettext-base git mtools parted python3 \
             quilt qemu-user-static reprepro sudo unzip git-buildpackage \
             pristine-tar sbuild schroot zstd \
             umoci skopeo \
@@ -138,11 +138,12 @@ RUN --mount=type=cache,target=/var/cache/apt,sharing=${CACHE_SHARING} \
     apt-get install --no-install-recommends -y \
         gawk wget git diffstat unzip texinfo \
         gcc build-essential chrpath socat cpio python3 python3-pip python3-pexpect \
-        xz-utils debianutils iputils-ping python3-git python3-jinja2 libegl1-mesa libsdl1.2-dev \
-        pylint xterm python3-subunit mesa-common-dev zstd liblz4-tool && \
+        xz-utils debianutils iputils-ping python3-git python3-jinja2 libsdl1.2-dev \
+        pylint xterm python3-subunit zstd lz4 && \
     if [ "$TARGETPLATFORM" = "linux/amd64" ]; then \
         apt-get install --no-install-recommends -y gcc-multilib g++-multilib; \
     fi && \
+    ln -svf /usr/bin/lz4 /usr/bin/lz4c && \
     rm -rf /var/log/* /tmp/* /var/tmp/* /var/cache/ldconfig/aux-cache

 USER builder
```

Closing as I don't see as this could be `kas` issue.

closed 2025-02-07 by jetm

michaeladler, 2025-02-10:
> With `konsole` works. But I managed to make it work by installing `kitty-terminfo` and changing `TERM` to `xterm-kitty` inside of kas container.

Since `kitty-terminfo` is a dependency of `kitty`, I suspect that you downloaded a pre-built `kitty` binary from GitHub. It's generally recommended to use the Arch packages (if available) to avoid problems like this.

> `Dockerfile` changes for TERM change. After building with `./scripts/build-container.sh` and making `kas` to use that image with `export KAS_CONTAINER_IMAGE=ghcr.io/siemens/kas/kas:next`. Upgrading Debian is optional.

That shouldn't be necessary, as it's more of a workaround. Your goal should be to make the upstream container work on your system, as you may encounter more terminal-related problems with other TUIs (if you're a terminal user) in the long run.

> Closing as I don't see this as an `kas` issue.

Agreed :)