siemens/kas #143: Kas fails to run with latest update from release 4.7

johnvvang, 2025-03-03:
Hi, 
I have a codebuild project set up to build my image and it is currently just tracking the latest kas from master. It's set up to build on a daily basis and it's been working fine until 3 days ago my build failed at running `kas build image.yaml`. The specific error I got is:
> [Container] 2025/03/03 16:01:28.607261 Running command kas build image.yaml
> /codebuild/output/tmp/script.sh: 4: kas: Permission denied

I believe the issue started with https://github.com/siemens/kas/compare/59e55e4f4e38d659e04de0018c9e5671a2a2fedb...2694ad64bfa258e8fbdc984cde41ff37f35e9fd7
Please let me know there's anything I should note before using the latest kas commits. 

Cheers,
John

jan-kiszka, 2025-03-04:
Thanks for tracking regressions and reporting! Can you share more about what you are testing and how?

The referenced changes primarily bring dropping of Python < 3.9. Is your host affected by that? We have also dropped some workaround for older Python (8a2296b3a74bd0938364aa8acefda00ad19cb113). Those changes come closest right to playing a role, but the key is understanding what is actually your test setup.

johnvvang, 2025-03-05:
We have a AWS codebuild project defined in our pipeline. The codebuild project spins up a Linux EC2 container that hosts the image build. The process contains the following steps:
1. Set up the host environment - e.g. installing packages such as debootstrap, tree, git, python3, etc. We don't specify version on these packages so it would be whatever the latest when codebuild project is executed. 
2. Retrieve kas from https://github.com/siemens/kas.git
3. Install kas via pip3
4. Retrieve our own build package (we use ISAR - great job on that one!)  
5. Run kas build xxxx.yaml to start the build
This process has been working for a while and only started to break till last Saturday. And I can confirm by using kas 4.6 solved the issue.

jan-kiszka, 2025-03-05:
"And I can confirm by using kas 4.6 solved the issue." - did you mean 4.7?

Anyway, we still need more in order to help you. Options:
- you share the code/scripts that you run inside EC2 so that we can reproduce the issue
- if that is not possible, you try to reduce / rewrite your test case to something that is shareable
- if that fails, you try to debug the issue in your environment, and we can only support with suggestions based on what you report along the way