siemens/kas #146: github action build fails with

attina, 2025-03-24:
I want to build my layers in github action with the following workflow configure
```
name: Manual Yocto Build
on:
  workflow_dispatch:
    inputs:
      config:
        description: 'Config file path (e.g., kas/rockchip/scb600-openwrt.yml)'
        required: true
        default: 'kas/rockchip/scb600-openwrt.yml'
      clean_build:
        description: 'Perform full clean rebuild'
        type: boolean
        default: false

env:
  KAS_BUILD_DIR: build
  KAS_LAYER_DIR: layers

jobs:
  yocto-build:
    runs-on: ubuntu-latest
    container:
      image: ghcr.io/siemens/kas/kas-isar:latest
      options: --user root

    steps:
    - name: Checkout Repository
      uses: actions/checkout@v4
      with:
        submodules: 'recursive'
        fetch-depth: 0

    - name: Install System Dependencies
      run: |
        apt update
        apt install -y build-essential chrpath cpio gawk

    - name: Initialize Build Environment
      run: |
        mkdir -p ${{ env.KAS_BUILD_DIR }}

    - name: Clean Build Preparation
      if: ${{ inputs.clean_build }}
      run: |
        rm -rf ${{ env.KAS_BUILD_DIR }}/*

    - name: Execute Yocto Build
      run: |
        kas build ${{ inputs.config }}

    - name: Cache Management
      uses: actions/cache@v3
      with:
        path: |
          ${{ env.KAS_BUILD_DIR }}/downloads
          ${{ env.KAS_BUILD_DIR }}/sstate-cache
          ${{ env.KAS_LAYER_DIR }}
        key: ${{ runner.os }}-${{ hashFiles('external_layer.sha') }}-${{ hashFiles(inputs.config) }}

    - name: Package Artifacts
      uses: actions/upload-artifact@v4
      with:
        name: yocto-images
        path: ${{ env.KAS_BUILD_DIR }}/tmp/deploy/images/**/*
        retention-days: 3
        include-hidden-files: true
```
But the build failed with the following error:
```
2025-03-24 03:41:45 - ERROR    - ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
2025-03-24 03:41:45 - ERROR    - Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
2025-03-24 03:41:45 - ERROR    - Following is the list of potential problems / advisories:
2025-03-24 03:41:45 - ERROR    - 
2025-03-24 03:41:45 - ERROR    - Do not use Bitbake as root.
2025-03-24 03:41:45 - INFO     - 
2025-03-24 03:41:45 - INFO     - Summary: There was 1 ERROR message, returning a non-zero exit code.
2025-03-24 03:41:45 - ERROR    - Command "/__w/kas-build/kas-build/build$ /__w/kas-build/kas-build/layers/poky/bitbake/bin/bitbake -c build openwrt-image-minimal" failed
--- Error summary ---
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Do not use Bitbake as root.

2025-03-24 03:[41](https://github.com/attina/kas-build/actions/runs/14026738964/job/39266659551#step:7:42):45 - ERROR    - Command "/__w/kas-build/kas-build/layers/poky/bitbake/bin/bitbake -c build openwrt-image-minimal" failed with error 1
```
BTW, If I remove `      options: --user root` in my workflow configure file, the checkout stage will fail with permission problem.

attina, 2025-03-24:
workaround this problem by change
```
    - name: Execute Yocto Build
      run: |
        kas build ${{ inputs.config }}
```
to 
```
    - name: Execute Yocto Build
      run: |
        ./kas-container build ${{ inputs.config }}
```

closed 2025-03-24 by attina

jan-kiszka, 2025-03-24:
Yeah, more a github CI issue with yocto than something related to kas. If you tried yocto without kas, you would have run into the same problem.

attina, 2025-03-24:
> Yeah, more a github CI issue with yocto than something related to kas. If you tried yocto without kas, you would have run into the same problem.

@jan-kiszka You are absolutely right.

fmoessbauer, 2025-03-24:
Actually I'm wondering why you run the kas-isar container (`ghcr.io/siemens/kas/kas-isar:latest`) for a yocto build. For Yocto, use `ghcr.io/siemens/kas/kas:latest` and run rootless. The `kas-container` script can be used in both cases and takes care of selecting the correct container.