siemens/kas #150: Command "/tmp/xxxx/get_bb_env /build" failed with error 1

ysun, 2025-04-10:
Hi Sir,

Could you please help check why I'm encountering an error when using kas-container?
The command works fine with version 3.1, but it fails with any newer version.

```
KAS_ALLOW_ROOT=yes KAS_CONTAINER_IMAGE=ghcr.io/siemens/kas/kas-isar:latest ./kas-container --git-credential-store .git-credentials shell debian-sid.yml
2025-04-10 06:17:04 - INFO     - kas 4.7 started
2025-04-10 06:17:04 - WARNING  - Falling back to file-relative addressing of local include "xxxx.yml"
2025-04-10 06:17:04 - WARNING  - Update your layer to repo-relative addressing to avoid this warning
2025-04-10 06:17:04 - INFO     - Using /repo as root for repository xxxx
2025-04-10 06:17:04 - WARNING  - Using deprecated refspec for repository "isar". You should migrate to commit/tag/branch.
2025-04-10 06:17:04 - INFO     - Repository isar already contains 6c86d873189127592b8baa6f9fa7cc741c7f1e8d as commit
2025-04-10 06:17:04 - INFO     - Repository isar checked out to 6c86d873189127592b8baa6f9fa7cc741c7f1e8d
2025-04-10 06:17:04 - ERROR    - Command "/tmp/tmp168_d6t8/get_bb_env /build" failed with error 1
```

fmoessbauer, 2025-04-10:
Hi, a couple of things:

- please run the command in debug mode: `kas-container -l debug ...`
- `KAS_ALLOW_ROOT` is discouraged. Better add your user to the docker group
- `kas-isar:latest` != `kas-isar:latest-release`
- please try without setting `KAS_CONTAINER_IMAGE`

ysun, 2025-04-10:
@fmoessbauer thanks for the reply. I'll try the 4 bullets and update soon.
Before that, I have some finding: The kas container can clone the the repo successfully, while other version of container failed due to the permission denied. I'm using exactly the same configures/parameters, why only v3.1 have permission access the folder /work

`
│➜  sdak git:(xxx) ✗ KAS_ALLOW_ROOT=yes KAS_CONTAINER_IMAGE=ghcr.io/siemens/kas/kas-isar:latest ./kas-container --git-credential-store .git-credentials shell kas/debian-sid.yml
│2025-04-10 07:42:21 - INFO     - kas 4.7 started
│2025-04-10 07:42:21 - INFO     - Using /repo as root for repository rv64-sw-build-setup
│2025-04-10 07:42:21 - WARNING  - Using deprecated refspec for repository "isar". You should migrate to commit/tag/branch.
│2025-04-10 07:42:21 - INFO     - Cloning repository isar
│2025-04-10 07:42:21 - ERROR    - Command "/work$ git clone -q https://github.com/ilbers/isar work/isar" failed
│--- Error summary ---
│fatal: could not create work tree dir '/work/isar': Permission denied
│
│2025-04-10 07:42:21 - ERROR    - fetch repos failed: error code 128
│➜  sdak git:(sifive-fpga-wip) ✗
│
│➜  sdak git:(xxx) ✗
│➜  sdak git:(xxx) ✗ KAS_ALLOW_ROOT=yes KAS_CONTAINER_IMAGE=ghcr.io/siemens/kas/kas-isar:3.1 ./kas-container --git-credential-store .git-credentials shell debian-sid.yml
│2025-04-10 07:42:31 - INFO     - kas 3.1 started
│2025-04-10 07:42:31 - INFO     - /repo/kas$ git rev-parse --show-toplevel
│2025-04-10 07:42:31 - INFO     - /repo/kas$ git rev-parse --show-toplevel
│2025-04-10 07:42:31 - INFO     - /repo$ git rev-parse --show-toplevel
│2025-04-10 07:42:31 - INFO     - Using /repo as root for repository rv64-sw-build-setup
│2025-04-10 07:42:31 - INFO     - /work$ git clone -q https://github.com/ilbers/isar /work/isar
│2025-04-10 07:42:36 - INFO     - Repository isar cloned
│2025-04-10 07:42:36 - INFO     - /repo$ git rev-parse --show-toplevel
│2025-04-10 07:42:36 - INFO     - Using /repo as root for repository rv64-sw-build-setup
│2025-04-10 07:42:36 - INFO     - /work/isar$ git status -s
│2025-04-10 07:42:36 - INFO     - /work/isar$ git rev-parse --verify -q origin/213b8f62fa74e87321f927fb6b4ed9db93e5cb37
│2025-04-10 07:42:36 - INFO     - /work/isar$ git checkout -q 213b8f62fa74e87321f927fb6b4ed9db93e5cb37
│2025-04-10 07:42:36 - INFO     - /repo$ git rev-parse --show-toplevel
│2025-04-10 07:42:36 - INFO     - Using /repo as root for repository rv64-sw-build-setup
│2025-04-10 07:42:36 - INFO     - /work/isar$ git checkout -q -B patched-213b8f62fa74e87321f927fb6b4ed9db93e5cb37
│2025-04-10 07:42:36 - INFO     - /repo$ git rev-parse --show-toplevel
│2025-04-10 07:42:36 - INFO     - Using /repo as root for repository rv64-sw-build-setup
│2025-04-10 07:42:36 - INFO     - /work/isar$ /tmp/tmpmd80k_h5/get_bb_env /build
│2025-04-10 07:42:36 - INFO     - /repo$ git rev-parse --show-toplevel
│2025-04-10 07:42:36 - INFO     - Using /repo as root for repository rv64-sw-build-setup
`

fmoessbauer, 2025-04-10:
The following line indicates that you messed up your workdir permissions:

`fatal: could not create work tree dir '/work/isar': Permission denied`

Please check the permissions of this dir on the host. Please also try to use a different workdir by setting `KAS_WORK_DIR=my-workdir`.

fmoessbauer, 2025-04-10:
> I’ve figured out that the issue was caused by the old version of kas-container — after updating Docker, it looks like kas-container also needs to be updated to stay compatible.

That's interesting. Do you still remember the exact versions of the container (ideally digest) and `kas-container` script? We at least do not have any know API breaks between 3.1 and 4.x.

> The failure appears to be caused by the incorrect URL:
http://archive.ubuntu.com/ubuntu/dists/noble-updates/main/binary-riscv64/Packages
This is for the host distro, so the riscv64 architecture shouldn't be there.

Please bring this topic up on the Isar ML (`isar-users@googlegroups.com`). This is not a kas issue anymore :)

ysun, 2025-04-10:
> That's interesting. Do you still remember the exact versions of the container (ideally digest) and `kas-container` script? We at least do not have any know API breaks between 3.1 and 4.x.

[t.patch](https://github.com/user-attachments/files/19685247/t.patch)
Here is the diff between my v3.1 and v4.7.

> Please bring this topic up on the Isar ML (`isar-users@googlegroups.com`). This is not a kas issue anymore :)
Ok, get it.

Thanks your great help!

closed 2025-04-13 by ysun