siemens/kas #162: Kas-container overwrites /usr/share/zoneinfo/Etc/UTC with local timezone

stickypencil, 2025-07-24:
Due to recent commit https://github.com/siemens/kas/commit/37836639578ae4701380fae1d2c4b9830eea30f5 that mounts local timezone at `/etc/localtime` in container which also coincidentally overrides the `/usr/share/zoneinfo/Etc/UTC` since its a linked in debian based containers.

### Steps to reproduce

Here`s the minimal set of steps to observe the issue

```
# podman run -it -v /etc/localtime:/etc/localtime:ro ghcr.io/siemens/kas/kas:4.8.1
# md5sum /etc/localtime
# md5sum /usr/share/zoneinfo/Etc/UTC
```

### Solution

We can just also mount zoneinfo from host( e.g. ` -v /usr/share/zoneinfo:/usr/share/zoneinfo:ro`), or we can just use the `TZ` enviroment. 
Probably will try to submit a patch later today

fmoessbauer, 2025-07-24:
Hi, that is indeed not intended. It would be great if you prepare a patch for this (please see `CONTRIBUTING.md` how to submit it).

In case we have the `TZ`, we don't need the mounts. If we don't have it (because the caller does not set it), we need both `/etc/localtime` and `/usr/share/zoneinfo`. I missed that in my original patch.

stickypencil, 2025-07-24:
Hmm, scratch the part about it being the container image, when i use minimalistic query like this
```
# podman run -it -e TZ=<whatever> ghcr.io/siemens/kas/kas:4.8.1
```
it works as expected, so it seems that something just interferes with it in the full one that `kas-container` generates

fmoessbauer, 2025-07-24:
> Good news is that podman has --tz flag that does the heavy lifting and sets the provided timezone correctly itself.

Does this flag also exist on docker and if so, which versions of podman and docker support it? IMHO that is the preferred version, but we also need to care about users running on old distros with old podman / docker versions.

> On a side note is there any particular reason why you get the timezone from timedatectl instead of just cat /etc/timezone in your example, seems just a bit strange

If `TZ` and `/etc/timezone` are the only sources, we can just read from there. Then how about:

```
TZ=${TZ:-$(cat /etc/timezone)}
```

> so it seems that something just interferes with it in the full one that kas-container generates

You can run with `kas-container -l debug` to see the exact command line that is passed to the container runtime.

stickypencil, 2025-07-24:
> You can run with kas-container -l debug to see the exact command line that is passed to the container runtime.

It was exactly what i was doing, but it turns out i am severly dain bramaged, kinda forgot that the query runs `kas shell` which doesn`t preserve the enviroment variables)

> Does this flag also exist on docker

Sadly, no. It is `podman` specific flag and it is indeed prefered since it alteres the `zoneinfo` file and creates the `localtime` symlink.

So if we go with the `TZ` route we need to pass it down to `kas shell` somehow, or we can bite the bullet and prepend command that will run symlink the `localtime` and write the `zoneinfo` to `zoneinfo`

fmoessbauer, 2025-07-29:
After thinking a bit about possible solutions (without the revert), I noticed that it is not too hard to recreate the timezone information in the entrypoint. Please have a look at the following patch, which should fix your issue: https://groups.google.com/g/kas-devel/c/rxNDa9bhSZM

stickypencil, 2025-07-29:
@fmoessbauer Yeah, this should probably do it. Though testing it may take some
time, since as i mentioned the yocto recipe that creates the headache can`t be
built explicitly due to ```ASSUME_PROVIDED``` thingy... If you know how to
override that it would be apreciated

On a side note, now that i think about it, another advantage of using
`timedatectl` instead of `/etc/timezone` is that it is guaranteed to atleast be
correct, where `/etc/timezone` could have been borked by the user at some point
with him being none the wiser(or never been setup correctly at all, since in
distros like arch or say gentoo user is expected to fill them in by hand).
Though i do wonder whether `timedatectl` output follows the `TZ` enviroment
variable format(e.g. can things like `UTC+3` be an output of it), or does it
always correspond to `/usr/share/zoneinfo/*` structure

commit bd33a67b (2025-08-07): kas-container: propagate host timezone via entrypoint

Bind-mounting /etc/timezone into the container overwrites the symlinks
target, as both docker and podman mount without the AT_SYMLINK_NOFOLLOW
flag.  This breaks yocto builds which rely on correct information in
/usr/share/zoneinfo/*. To fix this, we pass the host timezone via an
environment variable to the container and patch the timezone in the
entrypoint. By that, we also preserve the difference between the TZ
variable and the /etc/timezone, /etc/localtime data.

Fixes: 378366395 ("kas-container: propagate timezone information ...")
Clo [... cut]
closed 2025-08-11 by jan-kiszka