siemens/kas #165: global installation of 4.8 fails

kwavnet, 2025-08-19:
```shell
sudo pip3 install kas~=4.8
[sudo] password for user: 
Collecting kas~=4.8
  Downloading kas-4.8.2.tar.gz (79 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 79.4/79.4 kB 1.9 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package kas produced metadata for project name unknown. Fix your #egg=kas fragments.
Discarding https://files.pythonhosted.org/packages/06/e2/f4f02db163634d92da0010f7c5eb9c55c77e3803bf79857d4d3488ddbe6b/kas-4.8.2.tar.gz (from https://pypi.org/simple/kas/) (requires-python:>=3.9): Requested unknown from https://files.pythonhosted.org/packages/06/e2/f4f02db163634d92da0010f7c5eb9c55c77e3803bf79857d4d3488ddbe6b/kas-4.8.2.tar.gz has inconsistent name: filename has 'kas', but metadata has 'unknown'
  Downloading kas-4.8.1.tar.gz (79 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 79.2/79.2 kB 9.2 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package kas produced metadata for project name unknown. Fix your #egg=kas fragments.
Discarding https://files.pythonhosted.org/packages/cb/5c/fd6950ade37076e800232d0398ba7f63d5cd404af0b379ecdc9f407e7426/kas-4.8.1.tar.gz (from https://pypi.org/simple/kas/) (requires-python:>=3.9): Requested unknown from https://files.pythonhosted.org/packages/cb/5c/fd6950ade37076e800232d0398ba7f63d5cd404af0b379ecdc9f407e7426/kas-4.8.1.tar.gz has inconsistent name: filename has 'kas', but metadata has 'unknown'
  Downloading kas-4.8.tar.gz (79 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 79.2/79.2 kB 9.7 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package kas produced metadata for project name unknown. Fix your #egg=kas fragments.
Discarding https://files.pythonhosted.org/packages/54/67/46feb1786e3a1904c399351cf1e15f0e97eed80b1ef947c72df1f0d49b41/kas-4.8.tar.gz (from https://pypi.org/simple/kas/) (requires-python:>=3.9): Requested unknown from https://files.pythonhosted.org/packages/54/67/46feb1786e3a1904c399351cf1e15f0e97eed80b1ef947c72df1f0d49b41/kas-4.8.tar.gz has inconsistent name: filename has 'kas', but metadata has 'unknown'
ERROR: Could not find a version that satisfies the requirement kas~=4.8 (from versions: 0.11.0, 0.13.0, 0.14.0, 0.15.0, 0.16.1, 0.17.0, 0.18.0, 0.19.0, 0.20.0, 0.20.1, 1.0, 1.1, 2.0, 2.1, 2.1.1, 2.2, 2.3, 2.3.1, 2.3.2, 2.3.3, 2.4, 2.5, 2.6, 2.6.1, 2.6.2, 2.6.3, 3.0, 3.0.1, 3.0.2, 3.1, 3.2, 3.2.1, 3.2.2, 3.2.3, 3.3, 4.0, 4.1, 4.2, 4.3, 4.3.1, 4.3.2, 4.4, 4.5, 4.6, 4.7, 4.8, 4.8.1, 4.8.2)
ERROR: No matching distribution found for kas~=4.8

[notice] A new release of pip available: 22.2.1 -> 25.2
[notice] To update, run: python3 -m pip install --upgrade pip
```

installation with `--user` or in a venv works

Environment: Ubuntu 22.04, python 3.10
All releases in before 4.8 were working too

jan-kiszka, 2025-08-19:
We switch to pyproject.toml in 4.8. The above seems like a bug in that pip version or some related python package in Ubuntu. 24.04 works fine this way. Also Debian 11 and 12 are happily choosing 4.8.2 for the installation. Nor is any of those versions except for Ubuntu 22.04 issuing that warning.

kwavnet, 2025-09-03:
I can confirm that a forceful update of pip in the base system is resolving the issue

jan-kiszka, 2025-09-08:
Can we close this then?

closed 2025-09-13 by kwavnet