siemens/kas #166: KAS has no way to pass BITBAKEDIR to oe-init-build-env and fails with impossible-to-debug error

b4u-mw, 2025-09-01:
I was debugging the following error message which I got without any explanation with log-level debug:

```
2025-09-01 11:37:49 - DEBUG    - /work/layers/external/gwf-legacy-openembedded-core$ /tmp/tmphnmhi52t/get_bb_env /build
2025-09-01 11:37:49 - ERROR    - Command "/tmp/tmphnmhi52t/get_bb_env /build" failed with error 1
```

My setup contains the layers:

```
  bitbake:
    path: layers/external/gwf-legacy-bitbake
[...]
  openembedded-core:
    path: layers/external/gwf-legacy-openembedded-core
[...]
```

The chain-of-cause here is:
- kas finds the first occurence of oe-init-build-env over all the layers
- openembedded-core contains one
- if is called in get_bb_env with only one parameter: https://github.com/siemens/kas/blob/7ecc070d52c16c5aa20101b33da6e69b6f0738eb/kas/libkas.py#L294
- so the init script will use the default logic to find bitbake: on the level of the init script or one above

Issues:
- the actual error `(The bitbake directory ([...]/layers/external/bitbake) does not exist!` gets somehow hidden. This seems not to be caused by the redirect to /dev/null but elsewhere since the error is on stderr.
- To fix this, I'd need to set the 2nd parameter which is not a bug but a missing feature ;-)

jan-kiszka, 2025-09-01:
> That's not the issue, it's using stderr. I think the issue is in https://github.com/siemens/kas/blob/master/kas/libkas.py#L194 where the stderr of a failing command is silently ignored.

This line only swallows errors if the caller set `fail` to `False`. Did you check that as well? Or provide a test case so that we can cross-check what is happening.

micw, 2025-09-04:
Sure here's the _minimal_ reproducer ;-)

* `file: test.yaml`

```
header:
  version: 13

repos:
  test:
    path: layers/test
    layers: {}
```

* file: `layers/test/oe-init-build-env`
```
#!/bin/bash

echo "Output from init script"
echo "Error from init script" >&2

exit 1
```

* command: `kas --log-level=debug build test.yaml`
```
[...]
2025-09-04 13:16:40 - DEBUG    - execute repos_checkout
2025-09-04 13:16:40 - DEBUG    - execute repos_apply_patches
2025-09-04 13:16:40 - DEBUG    - execute setup_environ
2025-09-04 13:16:40 - DEBUG    - /home/tester/kas-166/layers/test$ /tmp/tmp842vyv5n/get_bb_env /home/tester/kas-166/build
2025-09-04 13:16:40 - ERROR    - Command "/tmp/tmp842vyv5n/get_bb_env /home/tester/kas-166/build" failed with error 1
```

* Expected
    * Should show at least "Error from init script" in the output (also without debug)
    * Should show "Output from init script" (at least with debug

jan-kiszka, 2025-09-08:
Thanks for the helpful test. I've pushed two patches to next which should address this. Just note that we are not dumping the script's stdout on error, just stderr then.

b4u-mw, 2025-09-08:
Makes sense. Stdout goes to /dev/null in the command call anyway ^^

commit 92adcc48 (2025-09-14): libkas: Dump output of init-build-env scripts at debug log-level

This allows to inspect the output of the oe/isar-init-build-env scripts
when turning on debug log-level which can be helpful for analyzing the
proper script behavior.

Closes #166
commit ca7a34fb (2025-09-14): libkas: Consolidate and refactor error reporting of run_cmd*

While run_cmd_async was dumping stderr on fail=True, run_cmd was not.
This aligns both which specifically helps debugging failing
init-build-env scripts.

It furthermore makes the error dump more compact. Error reports are
already clearly marked by not carrying the prefixes of kas-generated
outputs.

Closes #166
closed 2025-09-14 by jan-kiszka